clearpathrobotics/robot_upstart #15: [Question] Why execute with setuidgid

pando85, 2015-01-30:
Hi, I'm using this fantastic node with my robot.
First of all, thanks, I like it too much.
Why do you use the command *setuidgid* to launch the roslaunch?

I use a user with groups added like video and dialout to have access the camara of my device (raspberry pi camera) and serial port.
I have to add dialout to main group and then I give privileges to dialout to access camera with:
```bash
$ echo 'SUBSYSTEM=="vchiq",GROUP="dialout",MODE="0660"' > /etc/udev/rules.d/10-vchiq-permissions.rules
```
But I think that doing this is very bad habit. How could it be done better?

Why don't you use just the command *su*?

Thanks for your time,
Pando

mikepurvis, 2015-02-05:
The principal reason is that `su` strips environment vars. This is not a deal-breaker, but it requires some re-thinking in terms of how the startup script works— for example, perhaps running the privilege downgrade right off the bat, instead of at the end immediately before the `roslaunch

pando85, 2015-02-05:
Ok I understand.
I found that for upstart >= 1.4 you can [setuid](http://upstart.ubuntu.com/cookbook/#setuid) but I coudn't modify that with exit so I will use your script adding privileges to my group. 
Thanks.

closed 2015-02-05 by pando85

mikepurvis, 2015-02-05:
I would be open to a PR which changes robot_upstart to use that functionality of upstart and avoid the dependence on daemontools. However, this isn't really going to change the current state of affairs— you're still stuck manually specifying a single UID/GID.