clearpathrobotics/robot_upstart #17: Multiple errors on Ubuntu 14.04

mzidek, 2015-02-11:
I tried to run robot_upstart on my Odroid board with Ubuntu 14.04, but it seems to fail:
```
odroid@odroid:~/catkin_ws$ rosrun robot_upstart install my_qbo_startup/launch/default.launch
grep: Invalid range end
grep: Invalid range end
grep: Invalid range end
/opt/ros/indigo/lib/robot_upstart/install: line 136: source: /bin/true: cannot execute binary file
Installing odroidindigo job using network interface eth0.
/opt/ros/indigo/lib/robot_upstart/install: line 153: rosrun: command not found
/opt/ros/indigo/lib/robot_upstart/install: line 154: rosrun: command not found
/opt/ros/indigo/lib/robot_upstart/install: line 155: rosrun: command not found
/opt/ros/indigo/lib/robot_upstart/install: line 162: rospd: command not found
Adding files to odroidindigo job: odroidindigo/home/odroid/catkin_ws/devel/setup.bashmy_qbo_startup/launch/default.launch/*
/opt/ros/indigo/lib/robot_upstart/install: line 168: pushd: home/odroid/catkin_ws/devel/setup.bashmy_qbo_startup/launch/default.launch: No such file or directory
cp: omitting directory 'etc'
cp: omitting directory 'include'
cp: omitting directory 'lib'
cp: omitting directory 'share'
/opt/ros/indigo/lib/robot_upstart/install: line 170: popd: directory stack empty
/opt/ros/indigo/lib/robot_upstart/install: line 172: popd: directory stack empty
```

mikepurvis, 2015-02-12:
Ugh, sorry about this. I'm not even certain entirely what's going on, but this type of failure is what motivated the Python rewrite (#14). Unfortunately, version 0.1.1 of the package is not yet being built on the ARM buildfarm.

Could you try using the indigo-devel version on your board from source?

trainman419, 2015-02-12:
The ARM buildfarm runs on my desktop, and it usually takes about a week of solid run-time to do a set of builds (during which I can't use my desktop for anything else), so I don't do builds very frequently.

The ARM builds are integrated into the new ROS buildfarm and scheduled for deployment soon, so hopefully the updated version will roll out to ARM when the main farm starts doing ARM builds.

mzidek, 2015-02-16:
Running the current git version solved the issue, thank you.

closed 2015-02-16 by mzidek