clearpathrobotics/robot_upstart #18: Device access with the `--user` option

galou, 2015-02-13:
I may be wrong but there is a limitation with the `--user` option because of a "limitation" of `setuidgid`.

Here is my scenario: I wanted to run the rosnode as user `ros` which belongs to group `dialout` and need access to some devices (e.g. /dev/ttyUSB0). My devices have ownership root:dialout and Read/Write permission for user and group. This setup allows me to run manually all nodes with device access.
However, when using `robot_upstart`, because of `setuidgid`, all groups other than `ros` are removed for the user launching the nodes (as stated in the man page). This means that `sudo service myrobot start` will not work because `ros`, not belonging to `dialout` when launched this way, does not have access to the devices.

Is there a way (other than setting `--user root`) to work around this problem?

The documentation should be updated to warn users about this fact.

mikepurvis, 2015-02-13:
The [previous version of the wiki page had a section on this issue](http://wiki.ros.org/action/recall/robot_upstart?action=recall&rev=2#Permissions), which got removed in the transition to Sphinx docs. I would welcome a PR to add it back to the new docs.

The present state of affairs is that you must chown the port to the ros user or make it world-rw, both of which require a udev rule and are not super-desirable.

But I agree that this is not great. I would entertain proposals to move the user-switching dingus to use su instead of  setuidgid, but envvars don't persist in that scenario, so it's more than just changing the one line.

Alternatively, we could get away from having to fork off a separate process altogether by switching the startup script to use Python, and using `os.setuid` and `os.setgroups` to demote the entire process. This deserves some exploration, though I haven't got the bandwidth to look at it in the immediate term.

galou, 2015-02-13:
PR #19.

closed 2015-02-13 by galou

galou, 2015-02-13:
Isn't `envuidgid` meant to do what we would need?

mikepurvis, 2015-02-13:
Pretty sure envuidgid [still just has the one gid specified](http://cr.yp.to/daemontools/envuidgid.html), so the root issue remains present.

laszipt, 2020-02-09:
Yes, the **`envuidgid`** does it. I have modified the generated start script and it works like a charm:

`envuidgid userID roslaunch $LAUNCH_FILENAME & PID=$!`

kledom, 2020-03-17:
Does anything speak against using `envuidgid`? I was having the same issue and envuidgid also fixed it for me.