clearpathrobotics/robot_upstart #20: Debian Jessie uses systemd to replace the init system

Waterfox, 2015-02-18:
Debian Jessie uses systemd to replace the init system

After installing robot_upstart, add the following file: /usr/lib/systemd/system/<robot_upstart_service>.service

----------------------------------------------------
[Unit]
Description=robot upstart pibot

[Service]
Type=oneshot
ExecStart=/bin/sh -c "pibot-start"

[Install]
WantedBy=multi-user.target
---------------------------------------------------

Then run: $sudo systemctl enable <robot_upstart_service>.service

This is based on instructions from here:
http://unix.stackexchange.com/questions/47695/how-to-write-startup-script-for-systemd

It was tested on a Raspberry Pi with Raspbian Jessie and ROS Indigo

mikepurvis, 2015-02-18:
Thanks for this. Systemd is the obvious choise for a second [Provider](http://docs.ros.org/api/robot_upstart/html/providers.html) for robot_upstart. Can't say how quickly I'll get to it, but it's a helpful starting point for providing general support— which will obviously be necessary for ROS K, since that will be on Ubuntu 16.04, which will use systemd for init.

Based on how the `xxxx-start` script works at present, I think the Type field should possibly [actually be "forking"](http://www.freedesktop.org/software/systemd/man/systemd.service.html#Options), as the parent bash script spawns the actual roslaunch job. However, I've become less satisfied with that whole thing— I think a better approach would be for the wrapper script to be a Python affair which exec's itself into the roslaunch job. Anyhow... we'll see.

Either way, thanks!

mikepurvis, 2017-01-23:
We have systemd support now.

closed 2017-01-23 by mikepurvis