clearpathrobotics/robot_upstart #23: Custom job start templates or ability to add pre-launch scripts

SISegwayRmp, 2015-03-18:
I needed to run startup checks on the system before executing roslaunch. Ultimately I ended up modifying templates/job-start.em to include various subsystem startup checks, such as network ping and platform identification before executing roslaunch.

I also made a change to providers.py which allows the user to pass custom template filenames. Alternatively a change could be made to specify pre-roslaunch scripts to be executed as part of the upstart job, I decided against this because it may be useful to set where the functions are called when the service is started.

The question is should the entire package need to be forked for minor changes in the template or is there some suitable way to allow for customization of the templates?

If this isn't useful, feel free to disregard. I have solved the issue for myself but thought it might be useful for others.

mikepurvis, 2015-03-18:
A major potential use case is payload detection at startup— ie, checking for USB devices to be present, and setting envvars accordingly (which then impact launch and xacro).

The only present provision made for this type of thing is the `workspace_setup` parameter to Job, which defaults to `/opt/ros/DISTRO/setup.bash`. However, you can point it to a develspace somewhere, or you can point it to a wrapper.

On CPR platforms, we use `/etc/ros/setup.bash` for this purpose, to set envvars, do any detection stuff, etc. The only caveat is that we _also_ make that the default setup file sourced in any terminal, so anything run in there has to be reliable and very fast— no waiting on network pings or whatever.

A final note is that you can also use [catkin_add_env_hooks](http://docs.ros.org/indigo/api/catkin/html/dev_guide/generated_cmake_api.html#catkin-add-env-hooks) to run arbitrary scripts as part of any workspace's setup stage. Again, though, you can't do anything destructive or long-running.

Something really simple might be just having `job-start.em` set a var like `ROBOT_UPSTART_JOB=jobname`. Then you could use either of the methods above, but check for that var and only run your stuff when it's actually in the startup context.

Let me know your thoughts.

SISegwayRmp, 2015-03-18:
Your use case is exactly what motivated me to do this, I detect system components and then modify the default env-hooks that get loaded from a workspace (local or the installed ros package) using catkin_add_env_hooks as you suggest. This sets up the environment with any corrections before execution of roslaunch (which uses the variables that have been set).

I also use it to delay the startup service until all of the components are online and operational, sometimes the PC boots faster than everything else. I was also having trouble using it on a VM since the network interfaces come up slower than on a native system.

I started out down the same road as CPR (modifying the setup.bash for the ws). But quickly ran into exactly what you mentioned, it gets run everytime you open a terminal which posed quite a few issues. The checks I run need to finish before attempting to execute roslaunch. We check a variety of things like ensuring the platform is there, which platform it is, what sensors/controllers are attached, what the payload power SOC is, checking for system faults and system motion, setting up the power system WD to protect the PC, start the NTPD server from the system RTC, etc etc. All that needs to happen before ROS gets involved, to ensure safe startup and ensure reliability in system timing.

Being able to setup customized templates in the upstart job would really be the cleanest way, so that ROS is not involved until everything is checked and the environment is setup correctly. Otherwise I see a need for changes to code that gets executed by roslaunch in the upstart job to wait for a variable to get set or something. In our case that is quite a bit including released packages maintained by others.  

What I needed was to be able to run the checks and setup the environment, with no changes to released packages and without dependancy on upstart (ie be able to run the same code without upstart). 

Atleast that's my opinion for right now :) , what are your thoughts?