clearpathrobotics/robot_upstart #24: Launch files in /etc becoming stale

jjekircp, 2015-04-21:
I sometimes get bit by an inability to start my robot_upstart job or other failures when this sequence of events happens:

1) I check out a package that contains a launch file I want to start at boot
2) I run robot_upstart install, which copies the launch file to /etc
3) I update the package (apt-get upgrade or git pull+recompile) and end up with an updated launch file in the package directory
4) I try to launch again but use the out-of-date launch file in /etc, and my package doesn't function correctly.

The hack I used by hand to get around this was by putting a "stub" rosinstall in `/etc/ros/<distro>/<jobname>.d`, that contained an include statement pointing to the launch file I wanted to launch. That way there is no copy of the launch file that can become out of date.
(Note that I'm on hydro so I'm using version 0.0.7...to the best of my knowledge this behaviour hasn't changed in newer versions of the package, but I mention this just in case).

Would you consider in a future version using stub launchfiles instead of copying to avoid this issue?

Regards,

Jonathan

mikepurvis, 2015-04-21:
Another possibility (and previously suggested) is creating symlinks rather than copying launchfiles. The argument for copying is that they can be edited without modifying a package-controlled file, however, it may make sense to introduce a option which links instead of copying.

jjekircp, 2015-04-21:
I thought about using symlinks as well. Stub launchfiles have the neat effect of doing a "rospack find" when they are run, so the launch file will be found either from the binary package in /opt or the source package in the workspace robot_upstart is using as appropriate.

I find this helpful for my use case, which involves both switching frequently from source packages to binary packages and back as well as launch files which change frequently, but I also appreciate that this might be a very niche consideration.

mikepurvis, 2017-01-23:
Closing given the fix in #43.

closed 2017-01-23 by mikepurvis