clearpathrobotics/robot_upstart #25: "Unable to locate path ... in package ... . Installation aborted" on one system, works on another

aleksandaratanasov, 2015-07-17:
Hi!

I have 3 computers with ROS running on them. The **robot_upstart install** works on two of those (Ubuntu) and doesn't work on the third one (Debian Wheezy).

On the first two I have the cob_bringup package inside the cob_robot package (part of the Care-O-Bot project by the Fraunhofer IPA). There is a robot.launch directly inside the cob_bringup package. When I run **rosrun robot_upstart install cob_bringup/robot.launch** the installation scripts detect the path properly. The full path of the launch file I have is basically **/home/USER/catkin_ws/src/cob_robot/cob_bringup/robot.launch**

On the third computer where the script fails I decide to do some testing and also added the robot_upstart package there (same procedure 1:1 as with the other 2 computers). The structure of this workspace is even simpler. The full path of the launch file is **/home/USER/catkin_ws/src/listenerTalker/launch/talker.launch** with the difference that I have the launch file in the launch folder (as per the file structure conventions in ROS - put the launch files in the launch directory, which is located in the root directory of the package). The found_path variable inside the **install_script.py** however returns empty (btw I added a return clause in this statement where the path is checked since before that the script just continues and right after that IF-statement crashes with an list index out of range (in found_path[0]) exception.

I have no idea what is causing all this. I even moved the launch file to the root of my listenerTalker package (to mimic the file structure on the other two systems) but all in vain. Note that this is a valid launch file and package (tested with roslaunch).

Any suggestions as to why this is happening?

aleksandaratanasov, 2015-07-17:
Nevermind. My mistake. After all it was not a valid package. I forgot to put the catkin_package() inside the CMakeLists.txt. Funny that roslaunch didn't complain about...

closed 2015-07-17 by aleksandaratanasov

gavanderhoorn, 2015-07-17:
Just fyi: `roslaunch` does not use `CMakeLists.txt`, that is part of the build infrastructure of your package. The `package.xml` is used by `roslaunch` (actually, the underlying libraries use the information in `package.xml`).