clearpathrobotics/robot_upstart #29: Log becoming huge in /var/log/upstart

roelnoort, 2015-10-27:
# First of all a disclaimer
Not 100% sure it this is an issue, a feature request or me not completely understanding how robot_upstart works. Please forgive me if I am putting this message in the wrong place.

# Issue description
We are using robot_upstart for our robot and are very happy with it. We face an issue with an ever-growing log file though. At some point the log becomes so big (94 Gb) that the systems if ' full ' and becomes unresponsive even to ssh logins.

# Our configuration
The install statement that we use is:
rosrun robot_upstart install --job doorman --setup /etc/ros/setup.bash decos_doorman/launch/doorman.launch

It creates the following files:
/etc/init/doorman.conf
/etc/ros/indigo/doorman.d/.installed_files
/etc/ros/indigo/doorman.d/doorman.launch
/usr/sbin/doorman-start
/usr/sbin/doorman-stop

# Possible solution: log to /tmp rather than to /var/log/upstart
Options that I found are :
* setting the environment variable ROS_LOG_DIR to /tmp
* passing --logdir /tmp to install statement 
This then assumes that Ubuntu takes care of clearing /tmp whenever it feels it is needed.
I tried both of these options but still the logs were being created in /var/log/upstart only.

# Possible solution: switch logging off completely
I did not yet find an option to not create log files at all.
As we use the logging for our 'production' robot and not on our R&D robots this would be a fine solution

# Possible solution: auto cleanup old logs
Ok, now we are getting into the feature request area I think.
The idea could be to keep only today's log and discard of older logs automatically.

mikepurvis, 2015-10-27:
The line that causes the logging to /var/log/upstart is the `console log` line in the job configuration:

https://github.com/clearpathrobotics/robot_upstart/blob/e5917621ca8ecfecf8281f2f679f7620a287c213/templates/job.conf.em#L44

Docs here: http://upstart.ubuntu.com/cookbook/#console-log

You can manually remove that from the generated file if you like, or as you say, something should be added to warn about that file getting huge, and/or automatically rotating/removing it. I would welcome PRs to improve the situation for any of these— non-breaking changes can be backported to Indigo.

roelnoort, 2015-10-27:
Thanks a bunch Mike! Will try it out.

mikepurvis, 2015-10-27:
Looks like setting up logrotate is pretty easy, basically just create another file in `/etc/logrotate.d/`, eg, `/etc/logrotate.d/upstart-jobname` or similar.

Maybe this should just be done automatically as part of the upstart provider?

roelnoort, 2015-10-29:
I checked it out. Turns out that simply removing the line `console log` does not solve the problem; it continues to log in that case. Based on the link to the documentation you provided, I saw that `console none` might be the alternate statement to use to avoid logs to be created. Tested this out and it works as desired by me. Thanks a bunch Mike!

mikepurvis, 2017-01-23:
Closing as a special case of #4.

closed 2017-01-23 by mikepurvis