clearpathrobotics/robot_upstart #30: using robot_upstart seems fail

CrazyPopLin, 2015-11-02:
I installed robot_upstart from github into my odroid successfully. I am new to this ros package so I have some issue to use that.I follow the tutorial: [Usage](http://docs.ros.org/api/robot_upstart/html/#python-api).  
After i install and type the command `rosrun robot_upstart install my_ros_package/launch/test.launch`  in the terminal,then reboot.  
After that, I do `rostopic list`, it said **ERROR: Unable to communicate with master**  
But When do `roscor` , it said **roscore cannot run as another roscore/master is already running**. It seems that something works but I can't see.
Could you give me some suggestions to see what going on here. Thanks so much.

CrazyPopLin, 2015-11-04:
I can use uninstall now for last issue. but
 `rosrun robot_upstart install --job test1 --user odroid--setup /home/odroid/catkin_ws/devel/setup.bash --rosdistro indigo --master http://localhost:11311 --logdir /home/odroid/.ros/log obstacle_avoidance/launch/test.launch`
I don't know the network interface of ROS.
The **install** command works successful.
When I reboot and do `rostopic list`, The only **rostopic** i can see is `/rosout`. The **mavros** rostopic not shown which is supposed to shown if quad and odroid connect correct. Because I can see them just run `roslaunch obstacle_avoidance test.launch`.
I looked this [problem](http://answers.ros.org/question/207892/robot_upstart-could-not-open-port-devttyusb0/) in ros forum, it seems he has some issue happened with `/ttyUSB0` which is the same device i used.
Any one could give me some suggestions? Thanks.

taogashi, 2015-11-15:
me the same, wait for suggestion T_T

CrazyPopLin, 2015-11-15:
@taogashi ，I write my own upstart script suggested by @vooon. Follow [here](https://github.com/vooon/rarog/tree/master/rarog_configs/rpi2), just write your script like **rarog.conf**, dont forget include `your_usb.rules` in like **90-ttyAMA0.rules**. Let me know if I can help  :)

mikepurvis, 2015-11-17:
Hi folks— can you clarify whether the Odroid is running Ubuntu or some other OS? If you're running upstream Debian or something else based on a different init system, robot_upstart's default jobs will not work for you.

Note that #20 is tracking a systemd provider for robot_upstart.

CrazyPopLin, 2015-11-18:
hi @mikepurvis ，i am using odroid running Ubuntu 14.04.

closed 2015-12-07 by CrazyPopLin