clearpathrobotics/robot_upstart #32: robot_upstart not finding install_script

hemantrobodub, 2016-01-16:
I need to start ros launch file at machine startup .

We are using Ubilinux on Intel/Edison. We are using ROS:

    /rosdistro: indigo
    /rosversion: 1.11.13

I am trying to follow these instructions http://docs.ros.org/api/robot_upstart... I have downloaded code from https://github.com/clearpathrobotics/... branch indigo-devel 
Here is how my folder looks
hemant@127:~/catkin_ws/src/robot_upstart/src/robot_upstart$ ls -ltr 
total 36
-rw-rw-r-- 1 hemant hemant 2408 Dec 26 16:06 uninstall_script.py
-rw-rw-r-- 1 hemant hemant 6667 Dec 26 16:06 providers.py
-rw-rw-r-- 1 hemant hemant 9173 Dec 26 16:06 job.py
-rw-rw-r-- 1 hemant hemant 4814 Dec 26 16:06 install_script.py
-rw-rw-r-- 1 hemant hemant 1663 Dec 26 16:06 __init__.py
Following shows command I ran.
hemant@127:~/catkin_ws$ catkin_make
hemant@127:~/catkin_ws$ catkin_make install
hemant@127:~/catkin_ws$ source ~/catkin_ws/devel/setup.bash

Now creating a job

hemant@127:~/catkin_ws$ rosrun robot_upstart install myrobot mybotpackage/test.launch
Traceback (most recent call last):
  File "/home/hemant/catkin_ws/src/robot_upstart/scripts/install", line 29, in <module>
    from robot_upstart.install_script import main
ImportError: No module named robot_upstart.install_script
Question : Why it is not able to find robot_upstart.install_script

I have been told to inform here  and  it could be an issue
For more information see here

http://answers.ros.org/question/223129/robot_upstart-not-finding-install_script/
Thanks,

shoemakerlevy9, 2016-11-03:
Did you install from source or using apt? If you intalled with apt then under /opt/ros/indigo/lib/robot_upstart you should see a file named install. If you installed from source you could try installing using apt.

mikepurvis, 2017-01-23:
After you `source ~/catkin_ws/devel/setup.bash`, the module should be on your PYTHONPATH, eg:

```
$ source devel/setup.bash
$ echo $PYTHONPATH
/Users/mikepurvis/robot_upstart_ws/devel/lib/python2.7/site-packages:/opt/ros/indigo/lib/python2.7/site-packages
```

And that path should have the stub file which catkin generates to pass through to the src dir python modules, eg:

```
$ find /Users/mikepurvis/robot_upstart_ws/devel/lib/python2.7/site-packages
/Users/mikepurvis/robot_upstart_ws/devel/lib/python2.7/site-packages
/Users/mikepurvis/robot_upstart_ws/devel/lib/python2.7/site-packages/robot_upstart
/Users/mikepurvis/robot_upstart_ws/devel/lib/python2.7/site-packages/robot_upstart/__init__.py
/Users/mikepurvis/robot_upstart_ws/devel/lib/python2.7/site-packages/robot_upstart/__init__.pyc
```

Going to close this as it's super old and looks like a setup issue on your end.

closed 2017-01-23 by mikepurvis