clearpathrobotics/robot_upstart #37: Kinetic package?

chrisspen, 2016-06-27:
I noticed there's no ros-kinetic-robot-upstart package in the Kinetic release. This is unfortunate, because I found this project quite useful in previous releases. Is this project now dead?

mikepurvis, 2016-06-27:
Hi @chrisspen— nope, not dead. If you could test from source and confirm that the generated upstart job still works under Xenial's upstart-systemd adapter, that would definitely allow the release to proceed.

The ideal would be to supply a "provider" that creates a native systemd unit, and then some logic to intelligently select the default provider based on the current system's pid 1.

chrisspen, 2016-06-27:
I would be happy to test. How would I install from source? I've only ever used the package, and your help link in the README gives a 404 error. Is it enough to checkout and run `python setup.py install` or is there anything else I need to do?

http://docs.ros.org/api/robot_upstart/html/

mikepurvis, 2016-06-27:
Pull the package into your regular catkin workspace's src directory, build the workspace, and `source devel/setup.bash`. You should be able to use robot_upstart as if it was installed from apt.

I've corrected the URL in the readme.

chrisspen, 2016-07-03:
My target platform is a Raspberry Pi 3, and since there's no stable Xenial image for the Pi 3 yet, I can only test on Raspbian Jessie, but I believe it too has switched to systemd, so the test should be applicable.

For my test project "mybot", I have the file setup.bash that sources all relevant workspaces:

    source /opt/ros/kinetic/setup.bash
    source /usr/local/mybot/.env/bin/activate
    source /usr/local/mybot/src/overlay/devel/setup.bash
    source /usr/local/mybot/src/ros/devel/setup.bash

Installation of the robot_upstart package per your instructions went fine.

I was able to successfully install my upstart scrip by doing:

    sudo apt-get install upstart
    sudo su -
    source /usr/local/mybot/src/ros/setup.bash
    rosrun robot_upstart install --setup /usr/local/mybot/src/ros/setup.bash --user root --job mybot mybot/launch/all.launch

Note I had to explicitly install the `upstart` Raspbian package. Without this, the install reports success, but the service commands all fail. Once I installed the upstart package, then these commands all seemed to work:

    sudo service mybot status
    sudo service mybot stop
    sudo service mybot start
    sudo service mybot restart

However, the one error I noticed is that my service does not appear in the list generated by `sudo service --status-all`, as it did in Ubuntu 14.

mikepurvis, 2016-07-03:
We should support Xenial/Jessie without requiring that users of those platforms downgrade from systemd back to upstart, by creating a new subclass in this file:

https://github.com/clearpathrobotics/robot_upstart/blob/jade-devel/src/robot_upstart/providers.py#L75

And then ideally the following default param should be changed to some kind of smarter detection scheme:

https://github.com/clearpathrobotics/robot_upstart/blob/jade-devel/src/robot_upstart/job.py#L140

That said, detecting your init system is a hairy problem in itself: http://unix.stackexchange.com/questions/196166/how-to-find-out-if-a-system-uses-sysv-upstart-or-systemd-initsystem

Pinging @ibaranov-cp and @tonybaltovski who I know are interested in this as well.

chrisspen, 2016-08-14:
Ok, let me know if you need any more help testing, but this goes outside of my realm of expertise. In the meantime I'll have to drop this package and go with alternatives.

quattrinili, 2016-12-17:
I tried to use it on Ubuntu 16.04 with ROS kinetic, by using the current source code, trying to use the `provider` option. However, while launching the executable doesn't throw any error and everything seems to work fine, `service --status-all` doesn't show the new service in the list and indeed using  `service` to start the service does not work.

I don't have time right now to debug the problem, but when I have time, I will try to provide some input if the problem still persists.

paulbovbel, 2016-12-17:
Confirmed that under xenial, provider defaults to systemd, and installed job works as expected. Ready to go!

mikepurvis, 2016-12-18:
Fantastic, thanks @paulbovbel.

Should this be a new `kinetic-devel` branch and a 0.3.0 version number, or is it sufficient to tag 0.2.1 and release that to Jade and Kinetic together?

paulbovbel, 2016-12-19:
just a new tag makes sense..

mikepurvis, 2016-12-20:
https://github.com/ros/rosdistro/pull/13433
https://github.com/ros/rosdistro/pull/13434

closed 2016-12-20 by mikepurvis

paulbovbel, 2016-12-20:
Thanks Mike!

chrisspen, 2016-12-25:
Do you know when this will be available on the ROS repo? `apt-cache search ros-kinetic-robot-upstart` doesn't show anything.

mikepurvis, 2016-12-25:
@chrisspen It's in shadow now, so you can get it from there now, see: http://wiki.ros.org/ShadowRepository

See also the status page here: http://repositories.ros.org/status_page/ros_kinetic_default.html?q=robot_upstart

I'm not sure when the next sync to public will be, but the place for that discussions is here: https://discourse.ros.org/c/release