clearpathrobotics/robot_upstart #63: Upstart generates no logs

chrisspen, 2018-01-03:
Is there any special setup needed to get logging to work with robot_upstart?

In the previous version of robot_upstart, it worked perfectly, and logs were generated to the standard `~/.ros/log` directory. I recently upgraded to the most recent package in Ubuntu 16.04, installing my service with:

    source /usr/local/myrobot/src/ros/setup.bash; rosrun robot_upstart install --setup /usr/local/myrobot/src/ros/setup.bash --job myrobot --user ubuntu myrobot/launch/all.launch

My all.launch looks like:

    <launch>
        <include file="$(find myrobot)/launch/serial_head.launch" />
        <include file="$(find myrobot)/launch/serial_torso.launch" />
        <include file="$(find myrobot)/launch/sound.launch" />
        <include file="$(find myrobot)/launch/status.launch" />
        <include file="$(find myrobot)/launch/diagnostics_aggregator.launch" />
        <include file="$(find myrobot)/launch/robot_state_publisher.launch" />
    </launch>

Running `rostopic list` or `rosnode list` shows several things running, yet my `~/.ros/log` remains empty. This is especially frustrating since a few nodes aren't running, and I can't find any logs to shed light on the problem.

Is logging not supported via upstart, or does it require some special configuration?

wujiang, 2018-03-02:
Looks like it logs to `/tmp` by default.

mikepurvis, 2018-03-02:
Ubuntu 16.04 is now systemd-based, so there's no more `/var/log/upstart` directory. The stdout capture which used to go there is now available via `sudo journalctl -u myrobot`.

If you want to set `ROS_LOG_DIR` to something other than `/tmp`, pass `--logdir` to the install script or use `Job(log_path=...)`:

http://docs.ros.org/jade/api/robot_upstart/html/install.html
http://docs.ros.org/jade/api/robot_upstart/html/jobs.html