clearpathrobotics/robot_upstart #64: ERROR: Unable to communicate with master!

plnegre, 2018-03-29:
I'm trying a very simple setup to launch a roscore when robot boots. I followed these steps:

1. From a clean catkin workspace i cloned this repo: `cd ~/catkin_ws/src && git clone https://github.com/clearpathrobotics/robot_upstart.git`
2. I compiled the workspace `cd ~/catkin_ws && catkin_make`
3. I run the install script in the following way: `rosrun robot_upstart install --job roscore --interface enp60s0 --user my_user --setup /home/my_user/catkin_ws/devel/setup.bash --logdir /tmp robot_upstart/test/launch/a.launch`
4. I got this output:
```
/lib/systemd/systemd
Preparing to install files to the following paths:
  /etc/ros/kinetic/roscore.d/.installed_files
  /etc/ros/kinetic/roscore.d/a.launch
  /etc/systemd/system/multi-user.target.wants/roscore.service
  /lib/systemd/system/roscore.service
  /usr/sbin/roscore-start
  /usr/sbin/roscore-stop
Now calling: /usr/bin/sudo /home/my_user/catkin_ws/src/robot_upstart/scripts/mutate_files
[sudo] password for my_user: 
Filesystem operation succeeded.
** To complete installation please run the following command:
 sudo systemctl daemon-reload && sudo systemctl start roscore
```
5. I run: `sudo systemctl daemon-reload && sudo systemctl start roscore`
6. I rebooted the robot.
7. Once the robot is booted again, I opened a terminal and wrote: `rostopic list` and I got this error:
`ERROR: Unable to communicate with master!`

So, why the roscore is not launched at startup?

Also, there are no logs in `/tmp`.

mikepurvis, 2018-03-29:
Under systemd, you have to access the output of the process via the journal. Try:

```
sudo journalctl -u roscore
```

Alternatively, you can try launching the job in the foreground with `sudo roscore-start`, to observe it in real time. Either of these approaches should show you why roslaunch is not able to come up.

plnegre, 2018-03-29:
Ok, the problem was that I hadn't installed `daemontools`. After installing this everything worked right.

Thanks @mikepurvis  for your quick response!

mikepurvis, 2018-03-29:
Ah I see, you're running robot_upstart from source— if you install the debian package, it comes along as a dependency. Glad you're set now!

closed 2018-03-29 by mikepurvis