clearpathrobotics/robot_upstart #66: Log from service published to /var/log/syslog

maetulj, 2018-04-17:
I am experiencing a problem with robot upstart on ROS Kinetic. I install my bringup launch file and it automatically starts on system startup. I also set _--logdir_ and the logs are written there.
But the log messages from ROS (ROS_INFO, ROS_ERROR) are also written to /var/log/syslog. This does not happen when directly calling roslaunch, only when the service runs.

Example of a syslog line:
Apr 17 16:10:41 robot robot-start[31465]: #033[31m[ERROR] [1523974241.950112782]: Sick300 not connected, will try connecting again in 500 msec...#033[0m

This is problematic because syslog is quickly growing too big. 

I tried the solution from: [](https://github.com/clearpathrobotics/robot_upstart/issues/29)
but without any luck. 

Is there a solution for this? Thank you.

130s, 2019-09-22:
What version of `robot_upstart` are you using?

Although I might be wrong, I can't think of a reason that robot_upstart by default prints to `syslog` even after briefly looking at the code. Looks like it used to before https://github.com/clearpathrobotics/robot_upstart/pull/14

It might be also worth noting that by `--logdir` arg you can specify the output directory.

ahmetalperenvar, 2023-04-03:
I am facing the same issue right now. Maybe adding `StandartOutput=null` and `StandartError=null` lines to the `[Service]` section of the service file which is located at `/etc/systemd/system/` is one solution. I have TCP messages using localhost do you have a similar thing in your system @maetulj ? I am asking this to understand the cause of the problem.