clearpathrobotics/robot_upstart #70: service can be started up but rostopic cannot be echoed

syangav, 2018-06-17:
Hi! 

I followed the installation tips and got one executable file in /usr/sbin/ros. I tried to run it directly in foreground and it gave me everything the same as roslaunch which means all topics can be echoed. However, when I tried to do 'sudo service ros start', I can see rostopic list but rostopic echo returns nothing. 

I am using ubuntu 16.04, ros kinetic. I have set up the /etc/ros/setup.bash to make sure my environment variables are correct. I also chmod all files to be used to 777. 

Any help or suggestions are greatly appreciated! Thanks in advance.

eirikaso, 2018-10-02:
I have the same issue with empty topics. Did you find a solution?

msadowski, 2018-10-10:
I never run into such issue. It seems as if ROS_IP or ROS_MASTER_URI might not be set up correctly? Can you running the job with a --master flag?

eirikaso, 2018-10-10:
I am using this package to launch a usb camera driver, and it seemed to have some permission issues. I switched to root user "sudu su" and then ran "rosrun robot_upstart install myrobot_bringup/launch/base.launch". This seemed to solve the problem

gijskoning, 2019-01-09:
I fixed it by running rosrun robot_upstart install in "sudo su" and with the arguments --master and --interface. Without the arguments it would not work.