clearpathrobotics/robot_upstart #75: doesn't start my launch file after reboot

hansenmaster, 2018-11-01:
I used ROS kinetic on ubuntu 16.04 LTS PC, 

my mission to run this launch file after booting up (this on package turtlebot3_bringup and succeeded to run manually)
[image]

I did the 
rosrun robot_upstart install turtlebot3_bringup/launch/base.launch
[image]
This ask for  sudo systemctl daemon-reload && sudo systemctl start turtlebot3, i run the command

Then start the service sudo service turtlebot3 start ,, reboot, and nothing happen.

Could someone help me?

hoppss, 2019-01-02:
i also have similar result

rjosodtssp, 2019-08-23:
me too, have no idea

jeff-o, 2019-08-23:
Does ros start properly if you launch it manually?

sudo turtlebot-start (this will launch ros in the foreground) 

sudo systemctl start turtlebot3 (this will launch in the background)

If yes, then you probably have a permissions issue - a launch file in your turtlebot3.d folder is attempting to access something (usually a serial or USB device) that it does not have access to when triggered by a service.

If no, then pay close attention to any errors that are generated when you try to launch.

jash101, 2020-11-07:
I have similar issue and do try to access USB device. How should I solve the permission issue @jeff-o?

jeff-o, 2020-11-07:
First try launching ROS in the foreground using the "sudo turtlebot3-start" command. This will usually give you some clues about what node or nodes are crashing.

For USB devices, many require permission to be granted to the user in order to access them. If you know the location of the device (say, /dev/[device name]) you can give yourself temporary permission with the chmod command: "sudo chmod 666 /dev/[device name]". To make it so that you always have access on boot, you can use something called a udev rule. Many ros drivers will come with a udev rule, and all you have to do is move it to /etc/udev/rules.d. For drivers that don't have one, you can create a new rule. There are lots of great tutorials out there that are just a google away!

jash101, 2020-11-07:
Actually, I'd created udev rules previously, so I checked the logs at `sudo journalctl -u servicename`
and found there were problems with /tmp folder permissions. So I uninstalled the service and re-installed by giving a different ROS_LOG_DIR:
`rosrun robot_upstart install --logdir tempa package/launch/file.launch`

And it works fine now!
Thanks for your speedy reply @jeff-o