clearpathrobotics/robot_upstart #81: rosserial not working with service start

Druv98, 2019-05-14:
I have written a launch file to launch a publisher node and rosserial. When I launch the file using roslaunch it gets connected to the Arduino and the led starts blinking but when I use robot_upstart it didn't connect to Arduino. Although it launched the node which was supposed to be launched. So I can see the topics and nodes when I do rostopic/node list but the led on Arduino doesn't blink

Patrick-Lascombe, 2019-11-15:
Any update concerning this issue ?

mikepurvis, 2020-05-19:
This may be permissions. If you `sudo chmod 666 /dev/ttyACM0` (or whatever your Arduino is coming up as), does it then work? If so, you'll need to make sure that you either open up the permissions permanently for that device (using a `udev` rule) or that whatever user is running the service job is in the `dialout` group (and be careful because `setuidgid` doesn't carry over group memberships).

kledom, 2020-05-26:
I'm also having this issue (closely related to https://github.com/clearpathrobotics/robot_upstart/issues/18). I closed my pull request for this at https://github.com/clearpathrobotics/robot_upstart/pull/92 because this is probably not the right way to fix this. It worked as a workarround for me though.

The problem is as follows:
The setuidgid command used for dropping root privileges does not carry over the sublementary groups, meaning that even though the user was added to dialout, it will have no effect on robot_upstart, because the process in only executed as user robot:robot. I don't know the proper way to fix this. I couldn't find any easy method...it seems like envuidgid somehow works, although it shouldn't according to the documentation.

shu-qing, 2020-07-28:
All my usb virtual com port (/dev/ACM* with udev rules) can not launch properly when using robot_upstart.
But my usb serial port(/dev/ttyUSB*) works fine.