clearpathrobotics/robot_upstart #85: Release 0.3.1 or higher

130s, 2019-07-31:
Looks like only update since 0.3.0 is https://github.com/clearpathrobotics/robot_upstart/pull/73. But that's what I'd like to be merged in a Debian pkg.

If more updates being included is motivating for making a new release, someone from my team might be able to help reviews.

Thanks!

mikepurvis, 2019-08-01:
If you're willing to nominate additional changes to go in, that would indeed help motivate a release.

130s, 2019-10-12:
Looks like 5 issues have been closed since 0.3.0. Good enough to make a new release (^_^;) ?

130s, 2019-11-08:
@mikepurvis Would you need more change to make a release?

Or if it's the resource problem, I can offer making a release if that's okay for you (If you like that idea, then I will have to ask for any specifics about making a release since this is Clearpath' repo. I'll also need general ros release credential (write access on this and the release repo)).

130s, 2020-03-19:
@mikepurvis (or other maintainers if any), sorry for the noise, but would you mind taking small amount of time on this?

My team at Plus One Robotics has been cloning the source code because https://github.com/clearpathrobotics/robot_upstart/pull/73 is not yet available in binary.

As I said above, if you're too busy to make releases, I'm happy to help (I'm [14th frequent contributor to rosdistro](https://github.com/ros/rosdistro/graphs/contributors) so hopefully I'm qualifying enough as a co-release maintainer).

Thank you.

130s, 2020-12-09:
@mikepurvis Now I see Ubuntu 18.04 support  https://github.com/clearpathrobotics/robot_upstart/pull/88 merged in. Wouldn't that make a  valid cause for a new release for 1{6, 8}.04?

130s, 2021-03-03:
Saw https://github.com/ros/rosdistro/pull/28444 for Kinetic (and a few other PRs for other distros). Thanks @tonybaltovski.

closed 2021-03-03 by 130s