clearpathrobotics/robot_upstart #89: Incomplete log since Kinetic

FrancoisTeyssere, 2019-09-24:
Greetings, 

At the time I used this package with 14.04, Indigo and upstart, there was no issue getting a full log of every node in real time by using` tail -f ` on the myjob.log file. 

However, now that it's under 16.04 with Kinetic and systemd, I can't seem to find a viable option to obtain a full log, as if I were running roslaunch from a terminal. 

I tried `sudo journalctl -u myjob.service -f -n 50 -o short`, but only some of the messages are displayed here (I did not identify yet what decides which messages are shown or not). 

I also tried to set the log directory with --logdir, but I get one log file per node and rosout's logfile does not display all the messages either. 

Did anyone manage to have an aggregate view of all the log messages, as if launched from a terminal?

mikepurvis, 2019-09-24:
The journal for the generated unit contains all the stdout/stderr stuff emitted by `roslaunch`— exactly the content that used to go into the `var/log/upstart/[unit].log` file. Some other stuff may be ending up in `ROS_LOG_DIR` (by default `~/.ros/log`), but that was the case under both upstart and systemd.

paulbovbel, 2019-09-24:
journald may require some extra configuration, see the Storage and RateLimit* settings in particular.