clearpathrobotics/robot_upstart #94: Release for Noetic Ubuntu 20.04

Crowdedlight, 2020-06-11:
We are currently using your package as a great way to ensure our ros system starts upon boot. However we are currently moving to the new Noetic release, and noticed that there is no released ``ros-noetic-robot-upstart`` package. 

I can see there is a noetic branch. Whats is the timeline for getting a noetic version released? 

Can I manually compile and use it, by just adding the noetic branch files in a catkin_ws and build it? Or is there additional steps required?

mikaelarguedas, 2020-08-04:
+1 a python3 version released in ROS Noetic would be awesome

mmmherma, 2020-08-28:
Hello everyone:

Trying to use this node in Ubuntu 20 I have the following issue when running `rosrun robot_upstart install --job soar soar_gui/launch/soar_gui.launch`:

``Traceback (most recent call last):
  File "/home/aeorum/catkin_ws/devel/lib/robot_upstart/install", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/aeorum/catkin_ws/src/soar/robot_upstart/scripts/install", line 29, in <module>
    from robot_upstart.install_script import main
  File "/home/aeorum/catkin_ws/devel/lib/python3/dist-packages/robot_upstart/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 24, in <module>
ModuleNotFoundError: No module named 'job'

Does anybody experienced this problem?

Thank you in advance.

mikaelarguedas, 2020-08-28:
This package has not been fully ported to python 3 yet. 
You can try https://github.com/clearpathrobotics/robot_upstart/pull/96 that provides a python 3 port

mmmherma, 2020-08-31:
It worked like a charm!

Thank you so much!

mmmherma, 2020-08-31:
It worked like a charm! :)

Thank you very much!

JackBorer, 2020-10-29:
Is there any other reason that the current noetic branch would produce [this error](https://github.com/clearpathrobotics/robot_upstart/issues/94#issuecomment-682525608) on 20? Currently I am getting the same error,
`in <module> exec(__fh.read()) File "<string>", line 24, in <module> ModuleNotFoundError: No module named 'job'`
And I am unable to solve it using the current branch. Is there something else that is going wrong on my end?

mikaelarguedas, 2020-11-02:
The noetic branch is not compatible with Python 3 at the moment. You can try the branch from #96 that provides a python 3 port

mikaelarguedas, 2021-05-26:
The PR mentionned above has been merged and a new version in Noetic has been released. So I think this issue can be closed

closed 2021-05-26 by tonybaltovski