clearpathrobotics/robot_upstart #117: Starting service failed - cannot access '/etc/ros/galactic/turtlebot4.d/*.xacro' or '/etc/ros/galactic/turtlebot4.d/*.xacro'

jmm-slamcore, 2023-01-31:
I'm getting the following error when trying to run the service created by robot_upstart (installed via apt) in ROS2 Galactic.

```
 turtlebot4.service - "bringup turtlebot4"
     Loaded: loaded (/lib/systemd/system/turtlebot4.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-01-31 15:00:18 UTC; 9s ago
    Process: 3361 ExecStart=/usr/sbin/turtlebot4-start (code=exited, status=1/FAILURE)
   Main PID: 3361 (code=exited, status=1/FAILURE)

Jan 31 15:00:14 ganymede systemd[1]: Started "bringup turtlebot4".
Jan 31 15:00:15 ganymede root[3415]: turtlebot4: Launching ROS_HOSTNAME=ganymede, ROS_IP=, ROS_MASTER_URI=, ROS_HOME=/home/slamcore/.ros>
Jan 31 15:00:15 ganymede turtlebot4-start[3415]: <14>Jan 31 15:00:15 root: turtlebot4: Launching ROS_HOSTNAME=ganymede, ROS_IP=, ROS_MAS>
Jan 31 15:00:16 ganymede turtlebot4-start[3421]: ls: cannot access '/etc/ros/galactic/turtlebot4.d/*.xacro': No such file or directory
Jan 31 15:00:18 ganymede turtlebot4-start[3424]: ls: cannot access '/etc/ros/galactic/turtlebot4.d/*.launch.py': No such file or directo>
Jan 31 15:00:18 ganymede systemd[1]: turtlebot4.service: Main process exited, code=exited, status=1/FAILURE
Jan 31 15:00:18 ganymede systemd[1]: turtlebot4.service: Failed with result 'exit-code'
```

Is this a permissions issue?

tonybaltovski, 2023-02-01:
What happens when you run `/usr/sbin/turtlebot4-start`?

jmm-slamcore, 2023-02-01:
Updating my launchfile name from `turtlebot4_bringup_standard_launch.py` to `turtlebot4_bringup_standard.launch.py` seems to fix the issue.

```
ubuntu@ubuntu:~$ sudo slamcore_turtlebot4-start
<14>Feb  1 11:14:33 ubuntu: slamcore_turtlebot4: Using workspace setup file /home/ubuntu/slamcore-ros2-examples-priv/install/slamcore_ros2_turtlebot4_example/../setup.bash
<14>Feb  1 11:14:34 ubuntu: slamcore_turtlebot4: Launching ROS_HOSTNAME=ubuntu, ROS_IP=, ROS_MASTER_URI=, ROS_HOME=/home/ubuntu/.ros, ROS_LOG_DIR=/tmp
ls: cannot access '/etc/ros/galactic/slamcore_turtlebot4.d/*.xacro': No such file or directory
<14>Feb  1 11:14:36 ubuntu: slamcore_turtlebot4: Generated launchfile: /tmp/slamcore_turtlebot4.launch.py
<14>Feb  1 11:14:36 ubuntu: slamcore_turtlebot4: Started ros2 launch as background process, PID 8878, ROS_LOG_DIR=/tmp
[INFO] [launch]: All log files can be found below /tmp/2023-02-01-11-14-37-594767-ubuntu-8878
[INFO] [launch]: Default logging verbosity is set to INFO
...
```

I still get this error/warning though:
```
ls: cannot access '/etc/ros/galactic/slamcore_turtlebot4.d/*.xacro': No such file or directory
```
but the robot description seems to be loaded correctly.

tonybaltovski, 2023-02-03:
That warning shouldn't be an issue. Glad you got it working.

closed 2023-02-03 by tonybaltovski